CameraKit/camerakit-android #599: Sometime camera does not start, it opens on a dark black view, as if the camera is blocked

NehalBakr, 2020-01-08:
### Is this a bug report?

Camera usually works fine, but sometimes I open the camera kit component from my app, I find it opening on a dark black screen, I cannot capture a photo, check the attached screenshot
[image]

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

(Write your answer here.)

### Environment

(Please include the following information along with any other relevant environment details.)

CameraKit Version:

Android Device:

Android Version:

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

Ezugworie, 2020-03-18:
Yes please this is happening to me, what's the solution

bakonMa, 2020-03-19:
implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
 not 1.0.0-beta3.11

TigranSarkisian, 2020-04-07:
@bakonMa yes, thanks, works ok on 1.0.0-beta3.10 
black screen on 1.0.0-beta3.11

daisukikancolle, 2020-09-11:
@bakonMa yeah thanks, it works

husw, 2020-10-19:
> 实现'com.camerakit：camerakit：1.0.0-beta3.10'
> 而不是1.0.0-beta3.11

确实，使用1.0.0-beta3.10 可以解决此问题。

husw, 2020-10-19:
> implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
> not 1.0.0-beta3.11

确实，使用1.0.0-beta3.10 可以解决此问题。

PDymala, 2021-07-08:
3.10 instead od 3.11.

Best advice ever.