CameraKit/camerakit-android #601: Unable to record video on devices running Android 10

omid-io, 2020-01-23:
### Environment

CameraKit Version: 0.13.5

Android Device: Pixel 3 XL

Android Version: 10

### Steps to Reproduce

1. Add the camerakit library to any existing android project
2. set targetSDK to 29,  add CameraView to your layout + whatever camera settings you want + remaining setup like camera.start() etc. 
3. Try to record video and invoke the method call <your camera view instance>.captureVideo() passing in a CameraKitEventCallback<CameraKitVideo> callback. 
4. Invoke <your camera view instance>.stopVideo() and you will notice the callback you passed to captureVideo() never gets invoked by the camera kit library and a video file is not passed back. 

### Expected Behavior

A video file should be created and passed back to this method. 

### What is happening

The callback doesn't get invoked because devices running Android 10 use the new "scoped" storage by default and not the "legacy storage". Android R will be making it a requirement for using only the new "scoped" storage regardless of targetSDK. Libraries should check what type of storage is being used by makings calls to Environment.isExternalStorageLegacy(), which will return true if the app is in legacy mode, false otherwise, and use the appropriate file APIs.  After forking the library and debugging, I found this was the issue. What's happening under the hood is when we call captureVideo, the method captureVideo is called in the Camera1 class. A permission exception is bubbled up from the MediaRecorder.prepare() method which is called because the wrong file APIs are being used by the Camera1 class for an app using the new "scoped" storage. 

Temporary solution: apps can add `android:requestLegacyExternalStorage="true"` to their AndroidManifest to enforce using the old legacy storage and everything works fine. But this will break once Android R rolls out and makes it a requirement to using the new "scoped" storage. The camerakit library will break host apps at that point if they don't add support for reading/writing appropriately for the new "scoped" storage.

Himanshu507, 2020-02-27:
@omid-io  can you tell me how you record the video using this library because there is no documentation about how to record it and which callback is required. For this.

omid-io, 2020-02-27:
@Himanshu507 if you follow the "Steps to Reproduce"  in my first comment, it should guide you at a high level on the setup needed in order to record video via this library

closed 2022-01-28 by crocsandcoffee