CameraKit/camerakit-android #609: image wont saved on memory or gallery after capture listener run. error say open failed: EACCES (Permission denied)

AsyrafChingo, 2020-05-16:
### Is this a bug report?

(Yes)

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

(no)

### Environment

(Please include the following information along with any other relevant environment details.)

CameraKit Version:

Android Device: Samsung Galaxy A70

Android Version: 10 (API29)

### Steps to Reproduce

(Write your steps here:)

1. capture listener run from a button
2. no error, but no picture saved on memori or gallery. it say Exception 'open failed: EACCES (Permission denied)' on Android

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

alfonsotesone, 2020-06-09:
Unfortunately I have also encountered the same problem, is there no way to solve it?

alfonsotesone, 2020-06-11:
I managed to get around the problem. Essentially you need to use an AsyncTask to save the images. 😁

ashfaqa-blabs, 2020-07-20:
You need to provide the read and write access permission
   private void requestWriteStorageRuntimePermission() {
        if (ContextCompat.checkSelfPermission(this, Manifest.permission.WRITE_EXTERNAL_STORAGE) != PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(this,
                    new String[]{Manifest.permission.WRITE_EXTERNAL_STORAGE},
                    PERMISSIONS_REQUEST_STORAGE_CODE);
        } else {

        }
    }

    @Override
    public void onRequestPermissionsResult(int requestCode, @NonNull String[] permissions, @NonNull int[] grantResults) {
        switch (requestCode) {
            case PERMISSIONS_REQUEST_STORAGE_CODE: {
                // If request is cancelled, the result arrays are empty.
                if (grantResults.length == 1 && grantResults[0] == PackageManager.PERMISSION_GRANTED) {

                } else {
                    // permission denied,
                    ActivityCompat.requestPermissions(this,
                            new String[]{Manifest.permission.WRITE_EXTERNAL_STORAGE},
                            PERMISSIONS_REQUEST_STORAGE_CODE);
                }
                return;
            }
        }
    }