CameraKit/camerakit-android #611: error:  Camera1:Unable, waiting for picture to be taken

lirongzter, 2020-06-19:
### Is this a bug report?

(write your answer here)

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

(Write your answer here.)

### Environment

(Please include the following information along with any other relevant environment details.)

CameraKit Version:

Android Device:

Android Version:

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

TongReG, 2020-09-15:
Same error in API Level 23 too. Dont know how to fix it.

lirongzter, 2020-09-22:
maybe, call captureImage twice according to this error.