CameraKit/camerakit-android #612: Access denied finding property "vendor.camera.hal1.packagelist" in android

rajaniMakvana, 2020-06-22:
### Is this a bug report?

(write your answer here)

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

(Write your answer here.)

### Environment

(Please include the following information along with any other relevant environment details.)

CameraKit Version:

Android Device:

Android Version:

### Steps to Reproduce

(Write your steps here:)

1.
2.
3.

### Expected Behavior

(Write what you thought would happen.)

### Actual Behavior

(Write what happened. Add screenshots!)

### Reproducible Demo

(Paste the link to an example project and exact instructions to reproduce the issue.)

(Include your CameraKit setup and usage.)

Temirtator, 2020-12-28:
I have the same issue, but I want to know the reasons why this happens?

codewith-fun, 2022-09-09:
E/libc: Access denied finding property "vendor.camera.hal1.packagelist"

PID: 16628
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$4.done(AsyncTask.java:399)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)

PavlosIsaris, 2022-10-03:
Is there a solution to this issue?

nfuenma, 2022-12-05:
Has anyone found a solution for this?

Mnbbvvu, 2024-06-17:
**ghejw√ππnbue**

Mnbbvvu, 2024-06-17:
> 
> 
> ### Is this a bug report?
> yes
> 
> (write your answer here)
> 
> 
> 
> ### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?
> 
> (Write your answer here.)
> 
> ### Environment
> 
> (Please include the following information along with any other relevant environment details.)
> 
> CameraKit Version:
> com.wonderkiln:camerakit:0.13.4
> 
> Android Device:
> Redmi Note 7 Pro
> 
> Android Version:
> Android 9
> 
> ### Steps to Reproduce
> 
> 
> 
> (Write your steps here:)
> 
> 1.
> 2.
> 3.
> 
> ### Expected Behavior
> 
> 
> 
> (Write what you thought would happen.)
> 
> ### Actual Behavior
> 
> 
> 
> (Write what happened. Add screenshots!)
> 
> ### Reproducible Demo
> 
> 
> 
> (Paste the link to an example project and exact instructions to reproduce the issue.)
> 
> (Include your CameraKit setup and usage.)
> 
> 
>