CameraKit/camerakit-android #622: Camera not closing (onClosed in camera listener not called)

arulmani33, 2021-01-12:
### Environment: DEBUG

CameraKit Version:

Android Device: Samsung a10

Android Version: 10

### Steps to Reproduce

1. Open the camera and close normally with onBack Press
2. Open Second-time camera open fine but after close normally with onBack Press
3. Open Third-time camera not showing

### Logs:
2021-01-12 18:46:54.413 29694-29694/com.testing.test D/PhoneWindow: forceLight changed to true [] from com.android.internal.policy.PhoneWindow.updateForceLightNavigationBar:4238 com.android.internal.policy.DecorView.updateColorViews:1491 com.android.internal.policy.PhoneWindow.dispatchWindowAttributesChanged:3216 android.view.Window.setFlags:1148 com.android.internal.policy.PhoneWindow.generateLayout:2444 
2021-01-12 18:46:54.414 29694-29694/com.testing.test I/MultiWindowDecorSupport: updateCaptionType >> DecorView@564c3e6[], isFloating: false, isApplication: true, hasWindowDecorCaption: false, hasWindowControllerCallback: true
2021-01-12 18:46:54.414 29694-29694/com.testing.test D/MultiWindowDecorSupport: setCaptionType = 0, DecorView = DecorView@564c3e6[]
2021-01-12 18:46:54.446 29694-29694/com.testing.test I/testing.tes: Thread[1,tid=29694,Native,Thread*=0xeef76800,peer=0x72c9ef10,"main"] recursive attempt to load library "/data/app/com.testing.test-M_Iw-y7q8dCDsDVuoa7NSA==/lib/arm/libcamerakit.so"
2021-01-12 18:46:54.553 29694-29892/com.testing.test D/SDK:  pkg com.testing.test 4D669EB562B0A0B87E288AE180A72720
2021-01-12 18:46:54.580 29694-29694/com.testing.test D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 66667, 0,  
2021-01-12 18:46:54.581 29694-29894/com.testing.test I/CameraManagerGlobal: Connecting to camera service
2021-01-12 18:46:54.585 29694-29892/com.testing.test D/Test--linenum: 26
2021-01-12 18:46:54.585 29694-29894/com.testing.test D/VendorTagDescriptor: addVendorDescriptor: vendor tag id 3854507339 added
2021-01-12 18:46:54.590 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: setView = com.android.internal.policy.DecorView@564c3e6 TM=true MM=false
2021-01-12 18:46:54.592 29694-29713/com.testing.test I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_CLOSED for client com.testing.test API Level 2
2021-01-12 18:46:54.593 29694-29713/com.testing.test I/CameraManagerGlobal: Camera 1 facing CAMERA_FACING_FRONT state now CAMERA_STATE_CLOSED for client android.system API Level 2
2021-01-12 18:46:54.593 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: MSG_WINDOW_FOCUS_CHANGED 0 1
2021-01-12 18:46:54.593 29694-29694/com.testing.test D/InputMethodManager: prepareNavigationBarInfo() DecorView@c04f387[ActiveConversationActivity]
2021-01-12 18:46:54.593 29694-29694/com.testing.test D/InputMethodManager: getNavigationBarColor() -855310
2021-01-12 18:46:54.598 29694-29892/com.testing.test E/libscanner: >>>Version: 2017/01/05-340, WB_v340_CS_r<<<
2021-01-12 18:46:54.600 29694-29694/com.testing.test D/SurfaceView: onWindowVisibilityChanged(0) true com.camerakit.preview.CameraSurfaceView{6f3c6a6 V.E...... ......I. 0,0-0,0} of ViewRootImpl@3f56801[ImageScannerActivity]
2021-01-12 18:46:54.603 29694-29894/com.testing.test I/CameraManagerGlobal: postSingleUpdate device: camera id 0 status STATUS_PRESENT
2021-01-12 18:46:54.603 29694-29894/com.testing.test I/CameraManagerGlobal: postSingleUpdate device: camera id 1 status STATUS_PRESENT
2021-01-12 18:46:54.604 29694-29892/com.testing.test E/Secu-Check: pkg: com.testing.test
2021-01-12 18:46:54.607 29694-29892/com.testing.test E/Secu-Check: Sig: 4d669eb562b0a0b87e288ae180a72720 
2021-01-12 18:46:54.607 29694-29892/com.testing.test E/Secu-Check: pkg: com.testing.test
2021-01-12 18:46:54.610 29694-29892/com.testing.test E/Secu-Check: Sig: 4d669eb562b0a0b87e288ae180a72720 
2021-01-12 18:46:54.611 29694-29892/com.testing.test E/Secu-Check: ExpireDate 2021/6/30
2021-01-12 18:46:54.611 29694-29892/com.testing.test D/Test--Auth-code: 0
2021-01-12 18:46:54.611 29694-29892/com.testing.test D/Test---type_of_report: 0
2021-01-12 18:46:54.612 29694-29892/com.testing.test D/Test--Auth-Finish---code: 0
2021-01-12 18:46:54.613 29694-29892/com.testing.test D/ImageScannerActivity: code=0
2021-01-12 18:46:54.619 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=12 res=0x7 s={true 3089180672} ch=true
2021-01-12 18:46:54.628 29694-29713/com.testing.test I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_OPEN for client com.testing.test API Level 2
2021-01-12 18:46:54.633 29694-29694/com.testing.test D/SurfaceView: surfaceCreated 1 #8 com.camerakit.preview.CameraSurfaceView{6f3c6a6 V.E...... ......ID 0,0-720,1382}
2021-01-12 18:46:54.633 29694-29694/com.testing.test D/SurfaceView: surfaceChanged (720,1382) 1 #8 com.camerakit.preview.CameraSurfaceView{6f3c6a6 V.E...... ......ID 0,0-720,1382}
2021-01-12 18:46:54.641 29694-29890/com.testing.test D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-01-12 18:46:54.668 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)0 dur=11 res=0x3 s={true 3089180672} ch=false
2021-01-12 18:46:54.680 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
2021-01-12 18:46:54.680 29694-29694/com.testing.test D/InputMethodManager: prepareNavigationBarInfo() DecorView@564c3e6[ImageScannerActivity]
2021-01-12 18:46:54.680 29694-29694/com.testing.test D/InputMethodManager: getNavigationBarColor() -855310
2021-01-12 18:46:54.683 29694-29694/com.testing.test D/InputMethodManager: prepareNavigationBarInfo() DecorView@564c3e6[ImageScannerActivity]
2021-01-12 18:46:54.684 29694-29694/com.testing.test D/InputMethodManager: getNavigationBarColor() -855310
2021-01-12 18:46:54.684 29694-29694/com.testing.test V/InputMethodManager: Starting input: tba=com.testing.test ic=null mNaviBarColor -855310 mIsGetNaviBarColorSuccess true , NavVisible : true , NavTrans : false
2021-01-12 18:46:54.684 29694-29694/com.testing.test D/InputMethodManager: startInputInner - Id : 0
2021-01-12 18:46:54.684 29694-29694/com.testing.test I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2021-01-12 18:46:54.686 29694-29694/com.testing.test D/InputTransport: Input channel destroyed: 'ClientS', fd=189
2021-01-12 18:46:54.688 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: MSG_RESIZED_REPORT: frame=(0,0,720,1520) ci=(0,54,0,84) vi=(0,54,0,84) or=1
2021-01-12 18:46:54.707 29694-29694/com.testing.test D/ImageScannerActivity: cameraKit onOpened() called
2021-01-12 18:46:55.245 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)8 dur=17 res=0x5 s={false 0} ch=true
2021-01-12 18:46:55.248 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: stopped(true) old=false
2021-01-12 18:46:55.256 29694-29713/com.testing.test W/Gralloc3: allocator 3.x is not supported
2021-01-12 18:46:55.269 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: Relayout returned: old=(0,0,720,1520) new=(0,0,720,1520) req=(720,1520)8 dur=12 res=0x5 s={false 0} ch=false
2021-01-12 18:46:55.749 29694-29712/com.testing.test I/CameraManagerGlobal: Camera 0 facing CAMERA_FACING_BACK state now CAMERA_STATE_ACTIVE for client com.testing.test API Level 2
2021-01-12 18:47:02.182 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: ViewPostIme key 0
2021-01-12 18:47:02.199 29694-29694/com.testing.test D/ViewRootImpl@3f56801[ImageScannerActivity]: ViewPostIme key 1
2021-01-12 18:47:02.200 29694-29694/com.testing.test D/ImageScannerActivity: onBackPressed() called isProgressShowing - false
2021-01-12 18:47:02.216 29694-29694/com.testing.test D/SensorManager: unregisterListener ::   
2021-01-12 18:47:02.232 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: stopped(false) old=true
[... 165 lines cut ...]
2021-01-12 18:47:43.723 29694-29694/com.testing.test D/InputMethodManager: getNavigationBarColor() -855310
2021-01-12 18:47:43.725 29694-29694/com.testing.test D/ViewRootImpl@b629505[ActiveConversationActivity]: MSG_WINDOW_FOCUS_CHANGED 1 1
[... cut ...]
2021-01-12 18:51:35.817 29694-29738/com.testing.test I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
2021-01-12 18:51:35.817 29694-29738/com.testing.test I/System.out: (HTTPLog)-Static: isSBSettingEnabled false
2021-01-12 18:52:12.006 29694-29708/com.testing.test I/testing.tes: Background concurrent copying GC freed 324441(14MB) AllocSpace objects, 177(3588KB) LOS objects, 49% free, 20MB/40MB, paused 302us total 155.464ms
2021-01-12 18:52:12.020 29694-29710/com.testing.test W/System: A resource failed to call release.

chandra-meet, 2021-02-26:
did you find any solution for this??

BruceLii, 2022-12-15:
the  same problem in my demo ! carmera preview doesn`t work the second time after closed the camerakit activity ! and camera resouce samed not released   ! the camera  icon  still  there in status bar.