CameraKit/camerakit-android #639: Can't use CameraKit

Sagar0-0, 2022-08-15:
### Is this a bug report?

No

### Have you read the [Contributing Guidelines](CONTRIBUTING.md)?

Yes

### Environment

CameraKit Version: 1.0.0-beta3.11

Android Device: Realme RMX1831

Android Version: 9

### Steps to Reproduce

(Write your steps here:)

1. Add dependencies to the build.gradle (app level)
2. Add rules in proguard
3. Add xml code

### Expected Behavior

Use camera 

### Actual Behavior

Gradle sync issues with the following errors.
### In app level build.gradle
[image]

### In proguard-rules.pro
[image]

### In xml layout file
[image]

This was more of a setup problem than an actual project issue, was supposed to ask this thing in some community discussion section, but I found none. Also, Spectrum is also read-only now. I will request to have a community discussion platform for beginners.

Al-khateebBahaa, 2023-01-20:
You need to add **jcenter** to your Gradle.
Yes jcenter is down, but it won't work without adding jcenter 🤷🏻‍♂️

imufiid, 2023-07-18:
> but it won't work without a

I'm already jcenter in gradle, but it's the same not working