CameraKit/camerakit-android #643: Could not resolve com.camerakit:camerakit:1.0.0-beta3.10

hemant1109, 2023-07-24:
I am using this libs in my project and run Bitbucket pipeline. It gives me this error every time from 2 weeks.

Could not resolve com.camerakit:camerakit:1.0.0-beta3.10. > Could not get resource 'https://pay.cards/maven/com/camerakit/camerakit/1.0.0-beta3.10/camerakit-1.0.0-beta3.10.pom'. > Could not GET 'https://pay.cards/maven/com/camerakit/camerakit/1.0.0-beta3.10/camerakit-1.0.0-beta3.10.pom'. > The server may not support the client's requested TLS protocol versions: (TLSv1.2, TLSv1.3). You may need to configure the client to allow other protocols to be used. See: https://docs.gradle.org/7.2/userguide/build_environment.html#gradle_system_properties > Remote host terminated the handshake

Please help.

prikshitmaurya, 2023-08-01:
I'm also facing the same issue. can anyone please help here...

[image]

anurag-oneassist, 2023-08-01:
[image]

handirusli, 2023-08-05:
1. Visit : https://jitpack.io/
2. Type : https://github.com/CameraKit/camerakit-android on Git URL repo then click "Look Up"
3. Follow instruction on that page

Please let me know if it works.

anurag-oneassist, 2023-08-05:
cannot find it on any repo. 'jcenter' is down....

Resource missing. [HTTP GET: https://dl.google.com/dl/android/maven2/com/camerakit/camerakit/1.0.0-beta3.11/camerakit-1.0.0-beta3.11.pom]
Resource missing. [HTTP GET: https://jcenter.bintray.com/com/camerakit/camerakit/1.0.0-beta3.11/camerakit-1.0.0-beta3.11.pom]
Resource missing. [HTTP GET: https://repo.maven.apache.org/maven2/com/camerakit/camerakit/1.0.0-beta3.11/camerakit-1.0.0-beta3.11.pom]
Resource missing. [HTTP GET: https://www.jitpack.io/com/camerakit/camerakit/1.0.0-beta3.11/camerakit-1.0.0-beta3.11.pom]
Resource missing. [HTTP GET: https://dl.google.com/dl/android/maven2/com/camerakit/camerakit/1.0.0-beta3.11/camerakit-1.0.0-beta3.11.pom]

handirusli, 2023-08-05:
Hi @anurag-oneassist ,
Yes, i notice that jcenter down, but i found it in jitpack ( i'm using the older version )
have you change your dependencies to 
`dependencies {
	        implementation 'com.github.CameraKit:camerakit-android:v1.0.0-beta3.11'
	}` ?

pramahaditamaputra, 2023-08-07:
Hi @handirusli how about to solve the import com.jpegkit.Jpeg ?

fwallz, 2023-08-08:
@pramahaditamaputra have you tried the import below?

`implementation 'com.github.CameraKit:jpegkit-android:v0.2.2'`

hemant1109, 2023-08-09:
Hi @handirusli, thank you so much for your help. this is working fine for me.

klower, 2023-08-15:
Failed to resolve: com.camerakit:camerakit:v1.0.0-beta3.11

hemant1109, 2023-08-15:
https://github.com/CameraKit/camerakit-android/issues/643#issuecomment-1666638342

Please replace it with this suggested sdk. It will help.

projectdelta6, 2023-08-21:
Hi, just found this issue when updating an older project, our project was using v0.13.5 but that build failed on Jitpack, [Build Log Link](https://jitpack.io/com/github/CameraKit/camerakit-android/v0.13.5/build.log), I've set the app to use v0.13.4 for now 🤞🏼 hoping there are no issues as a result

hemant1109, 2023-08-23:
@handirusli 
your SDK is not capturing image. cameraKitView.captureImage(ImageCallback callback) is not calling callback.onImage().

hemant1109, 2023-08-23:
> Hi, just found this issue when updating an older project, our project was using v0.13.5 but that build failed on Jitpack, [Build Log Link](https://jitpack.io/com/github/CameraKit/camerakit-android/v0.13.5/build.log), I've set the app to use v0.13.4 for now 🤞🏼 hoping there are no issues as a result

Does this mean that the mentioned issue is resolved ?

projectdelta6, 2023-08-23:
> Does this mean that the mentioned issue is resolved ?

No the build is still failed on Jitpack for version 0.13.5 so that version is not available to use

hemant1109, 2023-08-23:
> No the build is still failed on Jitpack for version 0.13.5 so that version is not available to use

But I am using v1.0.0-beta3.11. So any news for it ?

larcho, 2023-09-01:
> @pramahaditamaputra have you tried the import below?
> 
> `implementation 'com.github.CameraKit:jpegkit-android:v0.2.2'`

Done this, but now I'm gettin:

`Didn't find class "com.jpegkit.Jpeg" on path: DexPathList[[dex file ...`

I feel like CameraKit needs to be updated as well for this to work?

koolfreak, 2023-09-08:
> @klower
> 
> [#643 (comment)](https://github.com/CameraKit/camerakit-android/issues/643#issuecomment-1666638342)
> 
> Please replace it with this suggested sdk. It will help.

Im having same issue, does anybody solve this? thanks

koolfreak, 2023-09-08:
> > @hemant1109 "your SDK is not capturing image. cameraKitView.captureImage(ImageCallback callback) is not calling callback.onImage()." Im having same issue, did you find solution for this issue? thanks
> 
> Unfortunately did not find any solution. So I am using the old lib and this lib is not working with the Pipeline.

unfortunately i dont have the old lib, can you share the old library? both for camerakit and jpegkit? thanks

larcho, 2023-09-08:
> > > > @hemant1109 "your SDK is not capturing image. cameraKitView.captureImage(ImageCallback callback) is not calling callback.onImage()." Im having same issue, did you find solution for this issue? thanks
> > > 
> > > 
> > > Unfortunately did not find any solution. So I am using the old lib and this lib is not working with the Pipeline.
> > 
> > 
> > unfortunately i dont have the old lib, can you share the old library? both for camerakit and jpegkit? thanks
> 
> Try this.
> 
> ```
> implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
> implementation 'com.camerakit:jpegkit:0.1.0'
> ```

This works for you because you still have the cached version stored in your ~/.gradle/ folder.

I can share the old libraries with you, I just don't know if that goes against any licensing rules.

hemant1109, 2023-09-08:
> This works for you because you still have the cached version stored in your ~/.gradle/ folder.
>
> I can share the old libraries with you, I just don't know if that goes against any licensing rules.

Yes I got it in the cache dir. let's find another solution/lib to overcome this.

larcho, 2023-09-08:
> Yes I got it in the cache dir. let's find another solution/lib to overcome this.

Reality is, this library is overall no longer being supported. The last release was made back in 2019. I briefly tried downloading the source code myself and building it but without success.

There are 2 options here:
1. Someone goes ahead and updates this library adding support to the newer **jpegkit** among other things, build a release and publish it to a none deprecated repository.
2. Look for an alternative to this library. It's overall a bad idea to use unsupported deprecated libraries to begin with.

You can bundle the old libraries using a local copy (gradle supports that), but that's an interim fix.

koolfreak, 2023-09-08:
thanks guys, looking forward who can share their cache/save old libraries. I need to build my old app with these libraries. If I ever had one i will put it on lib folder and commit to repostiory

larcho, 2023-09-08:
@koolfreak you can find me on Twitter with the same handle, I'll share the library with you there.

koolfreak, 2023-09-08:
> @koolfreak you can find me on Twitter with the same handle, I'll share the library with you there.

ok i start following you, me is @cybernetics_net. also can't seem to dm you

koolfreak, 2023-09-08:
@larcho thanks for sharing...now it works!

fwallz, 2023-09-13:
Hi @larcho , @koolfreak , is there any way you could share the files with me too? I started following you both on twitter but i am unable to DM you. Thanks.

saulopef, 2023-09-14:
Hi, still getting error on a legacy app? can some one share a solution that anyone in the community can use?

larcho, 2023-09-14:
> Hi @larcho , @koolfreak , is there any way you could share the files with me too? I started following you both on twitter but i am unable to DM you. Thanks.

Notice this library is under MIT license, so distribution is allowed:
https://drive.google.com/file/d/1zBZgcY72E3Bldj2z8JoFfnTjZyN1ntUv/view

fwallz, 2023-09-14:
Hi @larcho , @koolfreak no need for the files anymore. managed to make it work. Thanks anyway.

For others to make this work:

In build.gradle:

```
implementation 'com.github.CameraKit:camerakit-android:v1.0.0-beta3.11'
implementation 'org.jetbrains.kotlin:kotlin-stdlib-jdk7:1.3.41'
implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-android:1.0.0'
```
Then download the Source code for jpegkit-v0.1.0

`https://github.com/CameraKit/jpegkit-android/archive/refs/tags/v0.1.0.zip`

Extract, copy/paste folder jpegkit in the root folder of your project.

 In build.gradle:

`implementation project(':jpegkit')`

in settings.gradle:

`include ':app', ':jpegkit'`

Build and that should be it. Happy coding.

fwallz, 2023-09-14:
Thank you for your help. I also posted another fix for this if anyone need
it. Have great day

larcho, 2023-09-14:
Considering the huge amount of stars this project has, someone should try to update the library as a whole.

My Kotlin is so very rusty though, but I'll give it a shot.

saulopef, 2023-09-14:
> Hi @larcho , @koolfreak no need for the files anymore. managed to make it work. Thanks anyway.
>
> For others to make this work:
>
> In build.gradle:
>
> ```
> implementation [...]

hi @fwallz, tks for your solution, now I'm facing this error, do you solved it?
```
Unable to load class 'org.gradle.api.publication.maven.internal.MavenPomMetaInfoProvider'.

This is an unexpected error. Please file a bug containing the idea.log file.
```

fwallz, 2023-09-14:
This is how my build.gradle for jpegkit project looks like:

```
apply plugin: 'com.android.library'

android {
    compileSdkVersion 32
    defaultConfig {
        minSdkVersion 25
        targetSdkVersion 32
    }
    externalNativeBuild {
        cmake {
            path 'src/main/CMakeLists.txt'
        }
    }
}

dependencies {
    implementation 'com.android.support:exifinterface:27.1.0'
}
```

Basically remove from top:

```
plugins {
    id 'com.jfrog.bintray' version '1.7.3'
    id 'com.github.dcendents.android-maven' version '1.5'
} 
```

And everything below:

```
dependencies {
    implementation 'com.android.support:exifinterface:27.1.0'
}
```

koolfreak, 2023-09-21:
@fwallz  just chec my github...thanks for pulling that altogether

645148995, 2023-11-07:
maven {url 'https://maven.aliyun.com/repository/public/'}
        maven {url 'https://maven.aliyun.com/repository/google/'}
        maven {url 'https://maven.aliyun.com/repository/central/'}
        maven {url 'https://maven.aliyun.com/repository/gradle-plugin/'}  使用阿里云镜像解决这个问题

EdgarPozas, 2024-05-26:
> maven {url '[https://maven.aliyun.com/repository/public/'}](https://maven.aliyun.com/repository/public/'%7D) maven {url [...]

It works for me using `gradle 8.4` I just added the maven urls in `settings.gradle` and it looks like 
```gradle
pluginManagement {
    repositories {
        google()
        mavenCentral()
        gradlePluginPortal()
    }
}
dependencyResolutionManagement {
    repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
    repositories {
        google()
        mavenCentral()
        maven {url 'https://maven.aliyun.com/repository/public/'}
        maven {url 'https://maven.aliyun.com/repository/google/'}
        maven {url 'https://maven.aliyun.com/repository/central/'}
        maven {url 'https://maven.aliyun.com/repository/gradle-plugin/'}
    }
}

rootProject.name = "Project"
include ':app'

```
Root `build.gradle`
```gradle
// Top-level build file where you can add configuration options common to all sub-projects/modules.
plugins {
    id 'com.android.application' version '8.4.1' apply false
    id 'com.android.library' version '8.4.1' apply false
}
```
Then added the camerakit package in the project `build.gradle`
```gradle
plugins {
    id 'com.android.application'
}

android {
    namespace 'com.company.example'
    compileSdk 34

    defaultConfig {
        applicationId "com.company.example"
         minSdk 23
        targetSdk 34
        versionCode 1
        versionName "1.0"

        testInstrumentationRunner "androidx.test.runner.AndroidJUnitRunner"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android-optimize.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
    buildFeatures {
        mlModelBinding true
    }
}

dependencies {
    implementation 'com.camerakit:camerakit:1.0.0-beta3.10'
    implementation 'com.camerakit:jpegkit:0.1.0'
}

```