CameraKit/camerakit-android #646: Failed to resolve: com.wonderkiln:camerakit:0.13.5

YYKKKKXX, 2023-09-27:
I already added `dependencies {
    implementation 'com.wonderkiln:camerakit:0.13.5'
}` to `build.gradle(:app)`

But it shows 
[image]

And later I watched a guiding video about camerakit on youtube,  then I tried`repositories {
    maven { url 'https://dl.bintrary.com/camerakit/camerakit-android' }
}` to `build.gradle(:app)`, but there was still something wrong,  and it did not go like the guiding video.

Have I do anything wrong? And what should I do now?

Thanks for reading.

devahmedshendy, 2023-11-19:
@YYKKKKXX Were you able to solve this?

harshakakarla77, 2023-11-20:
@YYKKKKXX I'm facing the exact same issue. Any leads?

devahmedshendy, 2023-11-20:
@YYKKKKXX I got it working with beta3.10 of version 1, but not stable and sometimes work.

riyopay, 2023-11-26:
**Add it in your root build.gradle at the end of repositories:**

	dependencyResolutionManagement {
		repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
		repositories {
			mavenCentral()
			maven { url 'https://jitpack.io' }
		}
	}

**Step 2. Add the dependency**

	dependencies {
	        implementation 'com.github.CameraKit:camerakit-android:v0.13.5-SNAPSHOT'
	}

hosseinkaramii, 2023-12-22:
Still this problem exists...

weeklynote, 2023-12-22:
已收到，我会尽快回复你！
                     ——刘劲字

YYKKKKXX, 2024-03-23:
> @YYKKKKXX Were you able to solve this?

No, later I chose to use another open project named 'CameraView'.

YYKKKKXX, 2024-03-23:
> @YYKKKKXX I'm facing the exact same issue. Any leads?

Sorry, I didn't solve the problem, later I chose to use another open project named 'CameraView'.